pyr0ball/pyr0piezo #3: Arduino Code is too large for Atmega48 (88+ is ok)

pyr0ball, 2019-01-10:
Current sketch uses 5900 bytes, well over the 4kb limit of the mega48.

Refactor using low-level code will be required to fit the sketch on the smaller flash size, or switch to mega88+ may be required

teemuatlut, 2019-03-03:
Hey I saw your project on reddit and browsed this github page for a bit. Is this issue still ongoing for you?

I spent a few minutes making changes and with the serial stuff taken out, I got it to 2426B of progmem and 47 of sram.
I suppose it might be possible to include the serial parser as well considering an empty sketch takes 382B and with `Serial.begin(9600);` added it increased to 1480B.

pyr0ball, 2019-03-03:
I'm actually migrating over to the ATSAMD21E series of MCU's after the beta testing is complete, which has enough memory not to need optimization (though that's not the main reason I'm switching).

Also serial + i2c are going to be required services in the post-beta versions.

I'm trying to keep the code and designs as easy to work with as possible so that other community members can futz around with them.

I do appreciate the offer to help though!

pyr0ball, 2019-09-25:
Closing as the firmware has definitely reached a point where 4K will never be enough

closed 2019-09-25 by pyr0ball