pyr0ball/pyr0piezo #20: Firmware Rev.2.x.x, Analog readings being displayed as 1.00 consistently. Not changing with PWM adjustment

pyr0ball, 2019-03-08:
After fixing https://github.com/pyr0ball/pyr0piezo/issues/19 a new issue showed up, which is that the analog reading serial output is only showing 1.00 regardless of the actual output.

Probing shows 3.84v and 4.22v on V_FOLLOW and VComp respectively

Svatevit, 2019-03-08:
Try Serial.println(VCompRef,2);

foodbandlt, 2019-04-22:
Addressed in #25 if you want to re-enable those debug variables in the future.  Each line of Serial.println takes a crazy amount of flash (up to 4% each in some cases) on the ATmega88.  Crazy.

pyr0ball, 2019-04-23:
@foodbandlt actually could you take another look at it after the merge? Notepad++ on one of my systems was badly configured and may have overwritten some of the whitespace fixes you did, and I had to manually merge, so I could have accidentally deleted that part

closed 2019-06-27 by pyr0ball