pyr0ball/pyr0piezo #22: Configure MCU for i2c slave device

pyr0ball, 2019-03-09:
Setting the sensor up as an i2c slave will allow to accept input form 3d printer controller as a peripheral device without needing to tie up a UART connection

klcjr89, 2019-03-27:
Have you considered doing a Kickstarter? I really would love to buy one of these boards, as I don't have the time to get the PCBs made and gather the components. I think your product would be the best one of all of the auto bed leveling options out there.

pyr0ball, 2019-03-27:
@klcjr89

It's definitely on my radar, but I want to get the beta (Rev.2.x) finished and validated, and at least start on the firmware for the migration to ARM before I start any crowdfunding, as I want to be sure I can deliver a working product without that usual 3-years-later hooey that usually happens.

I also have a ton of documentation to do, but that's getting worked on in between other development stuff. 

Anyhoo, if you're interested in a beta unit, feel free to message me directly or you can post on the discord channel #beta-signup.

Being a beta tester does require some level of expertise and a willingness to put in a little work to iron out the fine details of the project, but if you're up to it, I'd be glad to send you one!

klcjr89, 2019-03-27:
I have alot of 3D printing experience, so I would be interested in a beta unit; I can't seem to be able to join the Discord however, it's not working. I recently launched a Kickstarter myself that ends today: https://www.kickstarter.com/projects/1536120962/pei-coated-flex-plate-system-for-3d-printers/

I think the piezo would be the best probe to use on my surface.

Taylor-Kun, 2019-03-27:
You did a great job on that kick-started video!

klcjr89, 2019-03-27:
@Taylor-Kun Thank you for that, but the credit goes to my videographer. He did a professional job.

pyr0ball, 2019-03-27:
@klcjr89 try this link: https://discord.gg/RSuKj7

looks like your kickstarter is doing exactly what I did on my CR-10 but with better magnets. Definitely a fantastic idea. I might even be able to integrate this sensor with your idea if cavities are created in the mount

pyr0ball, 2019-04-23:
Deleted off-topic comments

klcjr89, 2019-04-23:
☹️

pyr0ball, 2020-04-27:
Fixed in #70 and merged into `develop`

closed 2020-04-27 by pyr0ball