pyr0ball/pyr0piezo #46: [FR] PCB pin compatible with arduino nano

AlmightyFrog, 2019-10-14:
**Is your feature request related to a problem? Please describe.**
Custom tailored boards are hard to get.

**Describe the solution you'd like**
A solution would be to separate generic microcontroller part (e.g. use arduino nano) from specific part (LM2902 opamp).
This can be achieved by using DIP components on either breadboard or better simple PCB.

**Describe alternatives you've considered**
An alternative would be design single sided PCB with SOIC to be etched at home.

**Additional context**
Is there a plan or interest in doing such design?
How about current v2.x schematics, is it pretty much stable or subject to change?

pyr0ball, 2019-10-14:
@AlmightyFrog I'd be happy to put together a layout like this if you want to make your own. Is there a particular MCU dev board you have in mind?

The current methodology will likely stay the same going forward, but when I switch over to the ARM/CortexM0 MCU's the circuit will likely be a bit simplified as the SAMD21 MCU I plan to use has a number of hardware peripherals that can eliminate the need for as many external components.

AlmightyFrog, 2019-10-16:
I find this project apealing and would like to test it at some of my printers, but it would be hard to get the specialised boards which also can't be easily extended.

I've had a look at some easy and widely available boards and came to the conclusion, that the Arduino Nano (see features and schematics at https://store.arduino.cc/arduino-nano) would be most likely the best choice.

It is:
- easy to use and flash
- widespread platform
- small form factor
- cheap
- available in both 3v3 an 5v (as you already limit to 8 MHz)

If you compare the schematics, at Nano there is at PB5 an LED attached at the SCK pin but that does not harm and can be ignored.

Having already all pins exposed for easy access, i guess a PCB could spare the test points, also power circuit is already available and therefore obsolete.
Flashing is possible over ISP header or USB-Serial chip onboard, so also that is fine.

What do you think @pyr0ball?
I see some options which have all their own up and downsides:
1. For etching at home, using single sided PCB with big footprints (minimum 0805 SMD for resistors and capacitors and SOIC for chip) would be optimal. 
2. For getting people to start on their prototype board, maybe only a easy alignment of components and showing which pins to connect where would be helpful.
3. For getting produced shields for Nano, there the gold standard would be using double sided PCB as this is cheaply available.

Option 1 is my preference to get started.
Option 2 I myself would not use, as I fear issues due to bad contact points using jumper wires.
Option 3 is interesting, but maybe only worth it after some time when it is clear that the layout is really stable.

---
With regards to your switch over to e.g. SAMD21 I get the point of reducing external components, but with a controller like that you also go away from easy available hardware. I'd never had a look at your project if you would use complex circuits or harder to get controllers.
So it would be nice, if at least the AVR ATmega family still would keep at least maintenance support for some time.

pyr0ball, 2019-10-16:
@AlmightyFrog with regards to the switchover to SAMD21, there's several reasons for doing that beyond the reduction in components:
- Futureproofing. The AVR chips are probably going to be around for a while yet, but the ARM chips are getting a lot more popular and nearly as cheap. I ended up landing on the SAMD21 because it actually costs less than the 328pb I currently use in my custom boards.
- Ease of access is improving. SAMD21 is becoming just as easy to get as 328p boards. Look at the Arduino Zero for instance
- Faster processing. the SAMD21 has an internal 24mHz oscillator, which means a lot of aspects of the circuit work faster, which has a secondary effect of allowing me to use cheaper capacitors for the circuit (currently the most expensive passive component)

In regards to your thoughts on the dev-board compatible circuit, There's a couple of things I wanted clarification on:
- Did you want a design that can be attached as a hat for the Arduino Nano from the getgo, or just the Piezo circuit with the appropriate pins broken out into a header that can be connected to?
- With home-etched circuits, are there any caveats I need to be aware of? I've only ever designed with fab houses in mind.
- Do you have an idea of maximum footprint for the board outline? I have a bad habit of making everything tiny, so working to a minimum/maximum board size would be helpful in making a board that anyone can build easily

AlmightyFrog, 2019-10-21:
@pyr0ball 
Thanks for mentioning Arduino Zero, I was not aware about that one. Still more expensive than an "first gen" Arduino board, but i guess availability is then no issue.
Only I don't get why you then not go for something like a STM32. They're available in a wide range and are powerful, too.
Even the STM32F103 which you get on the "Blue Pill" boards already have an Cortex M3 @72 MHz (not sure how high you can go with the internal 8 MHz using PLL) and cost only a few bucks. There would also be with stduino an arduino port for it, but I can't tell how complete it is as I used it only for basic I/O so far.

It's a good question whether to go with an Arduino hat or "just" the piezo circuit.
I'd argue it depends on, if you want to try to spread people using it hat might be the way to go, but for flexibility and ease of use i guess second option makes more sense.

As I want to verify first, if your piezo solution would work for my printers second option would be more than enough. Guess main issue is to get the delicate analog section as stable as possible.

For both home etching and isolation routing you should try to not make too small traces. They shall not be too near and it is a no go to e.g. between 0805 pads pull a trace through.
PCB shall be in best case one sided, so you shall try to avoid vias. As that it not always possible it is best practice to only do pairs of vias which do not cross other when you do straight connecting them by (even unisolated) wire.
I'd draft you an example, but didn't use eagle since it was bought by autodesk.

pyr0ball, 2019-10-21:
@AlmightyFrog 

Re: SAMD21 vs STM32: For this project, the actual amount of processing power required is very low, since most of the MCU involvement is just balancing the circuit, so having an M3 would be waaaaay overkill. The SAMD21E that I plan to use was down in the ~$1.35 range when I started speccing it out (bare IC) compared to the STM's which were closer to $2.30+. There's a followup project I have that could utilize an M3/4, but for this M0/M0+ is perfectly fine.

I haven't officially called one or the other the "one" yet, so it's still up in the air.

Check out https://github.com/pyr0ball/pyr0piezo/commit/98cc387b20a6d52af0343cddd30effe4b91d37e1 || https://github.com/pyr0ball/pyr0piezo/tree/master/Eagle_Files/Pyr0-Piezo_DIY-Boards for the standalone piezo circuit I just added.

I will have to make a new version for home etching though.

re: Eagle, IIRC you can get a free license as long as you don't need more than one sheet and only make 2-layer boards. All of the designs I've made so far fit within the free license

AlmightyFrog, 2019-10-29:
I've got the free license version of eagle now and had a look at the DIY-Standalone board.

Looks promising, would be fine for fabrication but for home etching/isolation routing it is a game of luck whether the wires "through" the pads of a resistor work or not.
Also soldering (as there is no mask) is then harder. The only point that would not work out most likely is your VCOMP_PWM going through the trig led pads as it is not centered. And also clearance at C19 might be an issue.

It's been a long time since I used eagle, but i will try to recreate a layout for that PCB which fits my needs and test it out.

pyr0ball, 2019-10-29:
@AlmightyFrog I use a managed library that's attached to my account. It allows me to utilize the Push to Fusion 360 with part models as well as make edits to footprints and such.

I'll do some research on how to make that managed library accessible publicly