pyr0ball/pyr0piezo #47: [BUG] Ready light doesn't turn green upon power up unless piezo is manually tapped.

klcjr89, 2019-10-25:
**Describe the bug**
The ready light on the 2.1.0 beta board doesn't light up unless the Piezo element is first tapped.

**To Reproduce**
To reproduce the behavior, the printer is first turned on, the beta board then boots up and green light flashes, and then remains off until a tap on the piezo element by a finger causes the green light to finally light up and remain on and ready to use for a homing or probing procedure.

**Expected behavior**
I expect for the green light to turn on and remain solid upon powering up the printer.

**Printer Used:**
 - Manufacturer:
Creality
 - Model:
Ender-3
 - Control Board:
Azteeg X5 Mini V3 Ethernet
 - Firmware:
Klipper

**Pyr0-Piezo:**
2.1.0 beta board

pyr0ball, 2019-10-25:
Confirmed. This is behavior I've seen on all the 2.1.0 boards I put together but only became apparent in firmware v2.1.3+ IIRC.

@loredan or @foodbandlt just want you guys to be aware as well so if you come across anything in code that might explain that you can let me know. I'm assuming it's something to do with the setup function, *or* there may be a capacitance issue that needs correction in the startup

loredan, 2019-10-25:
I have encountered this problem, but not sure about the cause yet

As a side note, we really should set up a naming convention for these LEDs, so we are not calling them "green" or "ready", but rather something like "power", "warning", "trigger" and "inverted trigger". It's especially important in case of mismatched colors

pyr0ball, 2019-10-25:
@loredan that's especially true as I mixed two batches of RGB 0606's in my container and apparently the colors are rearranged between them.

klcjr89, 2019-10-25:
I called it a ready light since the silksceen layer appears to show as RDY.

pyr0ball, 2019-10-25:
@klcjr89 That's exactly right, but each of the RGB LED's channels correspond to a different function.

At the moment on the Rev.2.1.0's that got sent out:
Red= Power (Vcc input is high)
Green= Calibration Alert
Blue=Trigger (which appears to not be working correctly. Fixing it in 2.2.0)

klcjr89, 2019-10-25:
Oh! had no idea there was a blue led in there too; nice!

pyr0ball, 2019-10-26:
@klcjr89 ok I figured out the issue

Back in v1.x.x firmware, I wasn't using a task scheduler and the reset of the trigger was handled by the main loop. Now it's handled by the interrupt function. 

Here's the fun bit. on boot, the trigger LED blinks several times in order to indicate it's starting up, and it's done via inverting a pin by a count which is doubled in the setup function, which makes it always have an even number. What happens if you flip a switch twice? It goes on and stays off. So yeah I just need to add a +1 to the part of the setup code that doubles the blink count flip-flop and that should fix it.

pyr0ball, 2019-10-27:
fixed in https://github.com/pyr0ball/pyr0piezo/commit/f8efc3d192da328ebaa0f9038c133dedaf4f9d0a

closed 2019-10-27 by pyr0ball