pyr0ball/pyr0piezo #48: [FR] Detect if no piezo element is plugged in to the PCB header.

klcjr89, 2019-10-25:
**Is your feature request related to a problem? Please describe.**
Feature request is not directly related to problem #47, but it would potentially save people from crashes if a piezo wire gets broken or element damaged.

**Describe the solution you'd like**
Upon powerup, the piezo board can interrogate the PCB header where the piezo element plugs into to see if anything is attached. This would be done in a method that can be later reused, as described below.

Additionally, the MCU can poll every XX amount of seconds or (however long is determiend to be best) to verify that a piezo is still plugged in and that a wire or solder joint has not broken. This would cause the user to investigate their wiring when their firmware cannot home or probe the Z axis, thus eliminating a crash.

pyr0ball, 2019-10-25:
Detecting a piezo disk plugged in is very doable, but it may not be possible to detect whether the disk is damaged. Being able to halt the system if the piezo becomes unplugged is definitely a required safety feature I believe, however.

No polling should be required when using interrupt pins. I'll set up one of the unused pins as a pullup input, and have that get pulled to ground when the piezo is attached. If the piezo is unplugged, the pin will go high and can set an error state that will keep the system in a "triggered" state until the issue is fixed, which should prevent the printer from damaging itself

pyr0ball, 2019-10-25:
Created new board revision v2.2.x for this feature, as it requires changing a header and attaching a new pin to the MCU

pyr0ball, 2019-12-31:
added in a9d9288

closed 2019-12-31 by pyr0ball