pyr0ball/pyr0piezo #49: [BUG] Serial input appears to be ignored if VERBOSE is not defined

foodbandlt, 2019-10-26:
**Describe the bug**
Title

**To Reproduce**
* Comment VERBOSE out
* Try to send something over Serial In that would have some output, like CONFIG
* There is no response

**Expected behavior**
MCU should send something back over Serial Out, like the current config

**Pyr0-Piezo (please complete the following information):**
 - Pp Version: EasyPiezi
 - Board Revision: [e.g. Rev.2.1.2]
 - Firmware Version: 2.1.4.6

pyr0ball, 2019-10-26:
Confirmed, I've seen this behavior since 2.1.4, but assume it was due to my programmer being broken.

pyr0ball, 2019-10-28:
Only partially fixed, still missing serial feedback while typing

loredan, 2019-10-28:
I still think that feedback is not strictly needed, most serial terminals I've used have feedback built in, just disabled by default

pyr0ball, 2019-10-28:
It's different to every implementation I've ever used, but honestly I just can't figure out how it got disabled, even after reading through the code again

loredan, 2019-10-28:
Not sure it ever was there, at least I don't see anything resembling echo in code prior to my changes

pyr0ball, 2019-10-29:
@foodbandlt @loredan After our testing today, I've not been able to replicate the behavior with the `VERBOSE` flag. I'm assuming it was related to the VSCode environment settings I used when compiling and flashing the beta units I sent out. The issue with feedback is still present, but the board does accept serial input with or without the flag mentioned in this issue. Closing for now

commit f8efc3d1 (2019-10-26): Adds 1 to the blinkcount integer, trying to fix #49 #bugfix
closed 2019-10-29 by pyr0ball