pyr0ball/pyr0piezo #62: [MOUNT] ADIMLab Gantry-S

ETE-Design, 2019-11-25:
**Is your printer similar to any that currently already have support?**
Like the Ender-3 Just a smaller cover / fan also screws to hold it is on the top

**What mounting scheme do you plan to use?**
Choose between: On-Printhead

pyr0ball, 2019-12-02:
@ETE-Design Any chance there are sources or models for the gantry carriage?

If not, I'll need to get one or have you take measurements (or make a dummy model if you're capable)

ETE-Design, 2019-12-02:
Sorry, no models aviable :-( But can take pictures, and deliver all the messurements you need :-) BTW. Is it possible to get a sample, cause I have the only rights to sell the Gantry-S in EU :-)

pyr0ball, 2019-12-03:
@ETE-Design The boards are still very much in beta, so if you don't mind working with an unproven board, I can work that out with you. DM or @ me on the discord and I'll coordinate with you.

re: mount design, pics with measurements is a good start. Any chance we could reach out to the manufacturer for a 3D model? That would make the process a lot easier and faster.

pyr0ball, 2019-12-05:
First draft of the mount for this model has been posted: https://a360.co/34VN7gC