pyr0ball/pyr0piezo #66: [FR]Reprap 2 and 3 Configuration

Afteraffekt, 2020-01-29:
Rerap 2 and 3 firmware configuration data missing.

pyr0ball, 2020-01-29:
👍 Will get documentation written once I have a chance to review with one of the beta testers who uses RepRap

Tinchus2009, 2021-06-02:
Any update on this? I use reprap 3.X and I would like to have this docs

pyr0ball, 2021-06-02:
I don't currently have a reprap capable printer to test and make instructions with, but I'd be happy to include documentation if I can get some details about setup from someone who uses it more regularly.

One of my beta testers provided a little snip he used to set up his ezpz: 
```GCODE
;M558 P8 F4200 T7000 H10 I1                                  ; Z probe type to PIEZO and the dive 
```

gueee, 2023-01-21:
Here's my config running on the following hardware/firmware:
Board: Duet 2 WiFi (2WiFi)
Firmware: RepRapFirmware for Duet 2 WiFi/Ethernet 3.4.0 (2022-03-15)
Duet WiFi Server Version: 1.26
`
; Z-Probe
M574 Z1 S2                                         ; Set endstops controlled by probe
M558 P8 C"!zprobe.in" H3 F2600 T9000 	   	   ; Set Z probe type mini ir sensor
G31 P500 X0 Y0 Z0                       ; Set Z probe trigger value, offset and trigger height
M557 X25:275 Y25:275 S50			   ; Define mesh grid
`