pyr0ball/pyr0piezo #69: [MOUNT] Railcore II ZL/ZLT

EmilyBarbour, 2020-02-28:
**Is your printer similar to any that currently already have support?**
HEVO

**What mounting scheme do you plan to use?**
on print-head

**Additional context**
Corexy printer
Fusion model for printer on this page along with other info.  https://railcore.org/hardware/

pyr0ball, 2020-02-28:
Thanks for putting that request in! I've already started working on an under-bed solution for this printer. It will require adding a "padding" material between the bed plate and the Z Yokes, like a rubber or felt grommet, or a compact spring perhaps.

[image]
https://a360.co/39bEYqz

If desired, I can also work on an on-head version. Just let me know what works best for you

notacoat, 2020-02-28:
This works! Thanks!

pyr0ball, 2020-02-28:
I've added a rubber grommet as well as a small printed piece to assist with actuation:
[image]

[image]

pyr0ball, 2020-02-28:
@notacoat I realize you'll probably need boards to test this integration method. Are you comfortable coordinating over Discord for that?

Do you want to procure your own boards or have me send you some?

For this integration, you'll need 1x EasyPiezi V2.2.0 or later, and 3x of the SingleDisk Adapter boards.

I'm happy to keep this issue open til you've tested it, but you can close it any time if you feel like it's done

notacoat, 2020-02-29:
@pyr0ball hey, I’d love to help out w testing, but I’m super busy at the moment and won’t have time to at least for a couple weeks.  If you pop into the Railcore discord, there would be a lot of interest in, I bet

pyr0ball, 2020-03-05:
@notacoat awesome, thanks for the tip!