pyr0ball/pyr0piezo #79: [FR]Klipper-config.md Lost?

Zero-Grav, 2020-10-30:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

pyr0ball, 2020-10-30:
I need s bit more info to go on here. If you're requesting a firmware configuration page for Klipper, there's nothing special about it as of yet, just set the input pin and zero offsets. Once Klipper introduces menu options for configuring the sensor, and there's gcode around it, I'll flesh out a full page on it.

Zero-Grav, 2020-10-30:
Thanks for your answer.
So this thing can only be used as a Z endstop in the klipper？

pyr0ball, 2020-10-30:
It can be used as a probe as well, you just won't be able to configure the sensor's settings through Klipper till they add support for that.

This is veering off a bit into support territory, so if you need help with getting a piezo sensor set up, you should reach out through the Discord channel or the support email: support@pyroballpcbs.com.