pyr0ball/pyr0piezo #83: Board burned out

garyriet, 2021-04-08:
Good morning
I was working with my board and it blew out. Was running 9v thru it

pyr0ball, 2021-04-09:
@garyriet could you provide a little more detail about what happened and how?

garyriet, 2021-04-09:
I was testing the board connected to a buck converter set to 9v dc which was connected to my external power supply 13.8v dc.   All voltage measured using a fluke 87 multimeter. Connected board to power and to 4 27mm piezo discs. 
Turned power on and was testing for about 5min when I smelled burning, shut power off and found the smell was originating from the board.  Discounted board and measured power it was still at 9v. Reconnected board and it would not power on
Gary

pyr0ball, 2021-04-09:
Can I see a picture of the wiring harness (with pins annotated if possible), as well as a high-res photo of the top of the EzPz board?

garyriet, 2021-04-09:
[image]

garyriet, 2021-04-09:
[image]

garyriet, 2021-04-09:
[image]

garyriet, 2021-04-09:
[image]

pyr0ball, 2021-04-10:
Thanks for sending a bunch of angles, that really helps!

can I see where this is being plugged in on the controller side?
[image]

pyr0ball, 2021-04-16:
just checking in to see if you still needed help with this?

garyriet, 2021-04-16:
Yes I do.  Sorry was busy with work. 

Gary Rietmann
Cell: (609) 276-1559

pyr0ball, 2021-04-16:
@garyriet I've deleted your previous comment as it contained personal information. Please keep in mind that this is a public forum, and it's generally inadvisable to post contact information in a way that's visible to anyone (bot spammers included)

Can you read the above messages and get back to me about how the board was connected?

garyriet, 2021-04-16:
[image]

garyriet, 2021-04-16:
It was connected in place of the precision piezo board with DuPont connectors

pyr0ball, 2021-04-16:
I'm really sorry to keep asking,  but I need a better idea of what you connected and where at both ends.

A simple drawn wiring diagram would be helpful, or a complete picture of the entire connection chain.

garyriet, 2021-04-17:
Sure I'll do a drawing morning

garyriet, 2021-04-17:
[image][image]

garyriet, 2021-04-17:
Sorry I don't have Visio at home

pyr0ball, 2021-04-20:
That's perfectly alright @garyriet 

I'll try and reproduce the issue and let you know what I find out

pyr0ball, 2021-04-29:
Haven't been able to set up a rig to reproduce just yet, RealJob™ has been keeping me late a lot the last couple of weeks.

I've got an idea what might have caused it but need to try out your wiring scheme to confirm.

garyriet, 2021-05-30:
?? Any info?

pyr0ball, 2021-06-01:
Sorry for the delay, I haven't been able to spend any appreciable time in my lab since the pandemic began, and now that I can get back in, I've had a backlog of high-priority stuff keeping me busy.

Here's what I know so far: There's a red circle around the problem
[image]

What you can see here is a blowout on the GPIO region of the Atmel MCU. What this means is that too much current or voltage was applied to one of the sensor's sensing, or output pins. Determining the cause of this is usually difficult, but most often is caused by user error (something getting connected to the wrong pin).

In your case, there's  a possibility that an over-voltage potential was created with the wiring scheme you used. In most electronics, it's a good idea to connect the ground planes of all separated electrical devices together, to avoid this problem. From the wiring scheme you provided, the ground wire between your main power supply and the ground on the input to your controller could have caused the sensor board to act as a ground-return, blowing it out. This is still considered user error, but this is also something that can be guarded against in my design, which I didn't think to add protection for.
The wiring scheme you showed me is missing a wire (indicated by a red line here) between the sensor and the ground on the controller.
[image]

I haven't had a chance to test this theory yet, and need lab time to do so