pyr0ball/pyr0piezo #85: [MOUNT] Ultimaker 2+

svpcom, 2021-09-29:
**Is your printer similar to any that currently already have support?**
It similar to railcore-ii (core-xy geometry)

**What mounting scheme do you plan to use?**
I'll prefer Under-Bed, but I have some doubts about heat from bed (for ABS bed heated up to 100C when piezo sensor temperature limit < 60C) and vibration from bed when it do Z-moves.

**Any existing mounts with public links?**
N/A

**Describe alternatives you've considered**
Possible variant to mount sensor on print head, but it also has issues with temperature.

**Additional context**
Do anyone have success with installing piezo z-probe to Ultimaker2+ ?

pyr0ball, 2021-10-03:
If you can provide some CAD sources for the UM2 or your own printer, that will help speed the design process for this.

Underbed is how I would typically approach this kind of printer, and the lovely thing about Piezo probing, is you don't necessarily need the piezo element pressed up against the hot parts. As long as mechanical force is applied when the nozzle touches the bed, it should work. Even sharp z-movements should be tuned out by the signal processing circuitry, assuming you're getting a decent mechanical actuation of the piezo element.

Underbed has one big downside, which is that the center of the heated bed will be less sensitive, due to flex in between the mounting points. If you've added support such that the center of the bed is more rigid, this should be a non-issue. Alternatively, you can have piezo's mounted at either end of the X-axis gantry arm, or combine underbed with an on-head element.

Keep in mind, this sensor can work with any number of piezo's hooked up in parallel, and you only need the one EasyPiezi. However, the longer the wire from the piezo element is, the more electrical interference and parasitic inductance will affect the signal. 

Let me know what you think, and if you have any other thoughts or ideas~

svpcom, 2021-10-03:
CAD files (hardware design for UM2+ is open-source):  https://github.com/Ultimaker/Ultimaker2ExtendedPlus

The table is on three springs, so I'm thinking of placing the piezo sensors underneath. Will the sensor work if it experiences constant spring pressure? That is, a bolt will pass through the drilled sensor, which tightens the spring on which the edge of the table rests. 

When printing with abs plastic, the table temperature is about 100C. The table is thermally insulated in the bottom, but inside the closed volume the iron platform to which the table is attached (through 3 bolts and springs) will still heat up after a while. Will this affect the performance and / or lifespan of the sensors?

pyr0ball, 2021-10-04:
> Will the sensor work if it experiences constant spring pressure?

Yes assuming the polarity is correct, you can even pre-tension the piezo and pick up the "relief" motion if you hook it up backwards

> a bolt will pass through the drilled sensor

You can certainly do this and make it work, but I don't recommend using drilled discs. You'll get much more consistent results by bending the disk across it's diametric center by allowing one side of the disk to be free moving, and the other statically mounted. I build a little PCB that uses 12mm disks that produces the same signal to noise ratio as a 27mm disk. You can find those up on my site, or you can use the sources in the hardware section of this repository to send out to JLC or another fab house if you'd like to build your own.

Any design I do will likely not include drilled discs, unless specifically requested

> Will this affect the performance and / or lifespan of the sensors?

Yes and no. Piezo elements have a particular behavior when it comes to heat, which is that their sensitivity is reduced as more heat is applied. That signal degradation is partially elastic, partially plastic. An analogy that fits is bending a piece of wire. If you deflect a copper wire a small amount, it will return to its original shape, which is an "elastic deformation", but if you bend it further, it will stay in it's deflected state, which is a "plastic deformation". The piezo elements' signal output has a similar effect with heat. After a certain temperature, the signal suppression becomes permanent, but below that threshold, it's elastic. That threshold is different between all elements, so you'll have to test with the ones you've got to find out how much heat they can take. 

There's an upside to this! This sensor has the ability to increase it's input signal gain on the fly, so even if the signal from your piezo elements is suppressed, it can be boosted by the sensor to a certain extent. It's also possible to increase the gain beyond the 11x limit of the software, but that would require changing a resistor on the board. If your particular setup requires a higher input gain, and you don't want to do your own, I can rework a couple of units for you.

Another way to address that issue is to migrate the piezo elements to the underside of the springs rather than being pressed up against the bed itself. 

> CAD files (hardware design for UM2+ is open-source): https://github.com/Ultimaker/Ultimaker2ExtendedPlus

Ah thanks, I'd forgotten that! I'll take a peek when I've got some free time this week and let you know what I think

pyr0ball, 2021-10-04:
I had a look, and I'm fairly certain a technique like [this one](https://www.thingiverse.com/thing:4836261) will work for the UM2

I'll get something knocked together when time permits!

svpcom, 2021-10-04:
@pyr0ball What do you think about using of ceramic piezo rings like that https://aliexpress.com/item/32871471778.html instead of piezo disks? I'm planning to pass bolts with springs via them.

pyr0ball, 2021-10-05:
The mechanical quality factor on those might be an issue, as I believe these are primarily used for ultrasonic transducers. If I'm making a correct assumption (cus I've never actually tried one of these) it would be kinda like trying to turn a gear reduction box from the output side.

The listing does say it's capable of pressure and vibration sensing, so it's very possible that could work.

I could be completely wrong about that tho, and now I know these exist at that price point I'm definitely gonna grab some to test with! I'll have to see what kind of mechanical requirements these have for actuation. It's very possible that you could just stick these between the bed knurled nuts and the bed platform and you might get a signal if it works how they say.

If you've got an oscilloscope handy and would be up for trying out some tests with my guidance, I can probably give you a pretty solid idea of what they need to work.

pyr0ball, 2021-10-05:
@svpcom Here's a use-case very similar to the UM2 that's utilizing my SingleDisk adapters:
[image]
[image]
[image]

I don't think the UM2 has any other mounting holes adjacent to the bed leveling screws, so we may need to resort to an adhesive or something similar for that kind of mount

pyr0ball, 2021-10-06:
Started working on this, and it looks like there *are* mounting holes I can utilize on the UM2! What I don't know is whether they're threaded or not.
[image]

Can you confirm for me? If they're not threaded you'll likely need a nylock nut to keep them in place

svpcom, 2021-10-06:
They are not threaded. They are intended for the initial leveling of the table during assembly

pyr0ball, 2021-10-06:
do you think they could be used with a nylock nut on the other side?

Alternatively, I could design a mount intended to be used with some 3M adhesive tape (the wall mount kind) instead?