pyr0ball/pyr0piezo #86: [MOUNT]

calebols, 2022-03-29:
**Is your printer similar to any that currently already have support?**
Please list any similar printers that already have Piezo mounts
I couldn't find any when searching google or specifically Thingiverse

**What mounting scheme do you plan to use?**
Choose between: On-Printhead, Under-Bed, Gantry-Mount, or Other [if Other, please describe]
I'd like to mount this to the printhead but I'm open to suggestions.  I currently have a 3-point bed leveling system in place if that makes a difference.

**Any existing mounts with public links?**
Create a bullet-point list of any links to mounts similar to what you need, for example, from Thingiverse, TinkerCAD, GrabCAD, etc.
There are none that I could find.  I saw 1 for a hemera but for a completely different printer that used linear rails instead of the aluminum extrusions found on an ender.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
Other than going with a different ABL system, I'm not sure what alternatives there are.  The biggest "alternative" I've considered was finding a way to put the piezo's under the bed but I've not figured out how to use them with the bed springs as the bed springs have so much tension that I would think they would always be triggered. 

**Additional context**
Add any other context or screenshots about the feature request here.
So, I didn't see a place to specify above but I've got an original Ender 3 with an E3D hemera mount utilizing the new REVO nozzle ecosystem.  As mentioned above, I did switch out the 4-point bed leveling screws with 3-point bed leveling screws.  I don't think it will matter for the mounting but I've got an BTT SKR Mini e3 v3 board installed and I'm running Klipper on a Raspi 3b+.

Thank you in advance for any help!

pyr0ball, 2022-03-30:
With something like the Hemera, you'd probably be better off going with [the under-bed mount](https://www.thingiverse.com/thing:4836261) just due to the inertia involved with those extruder.

If you wanted it on-head still, I'd probably recommend switching to MGN rails to address the stability issues that might crop up

calebols, 2022-03-30:
Thank you for that...I'm not sure how I missed that link!  Unless it was because I had waaayyy too many tabs open while searching :).  I'll give the under-bed mounting a try and see how it goes.  Thanks again!