pyr0ball/pyr0piezo #90: doc-site down & some Q's

hannpet, 2024-09-25:
I stumbled upon your project, here is some feedback:

- https://docs.pyroballpcbs.com/ sadly seems to be down!
- Where can I buy those Boards? Maybe a link to PCBway or Aliexpress or similar?
- If I want to connect 4 Piezo-rings (underbed), can those be connected in parallel with a 10k Resistor each (like the PrecisionPiezo-Board does it), or is it necessary to use 4 distinct boards because there is more smartness involved?
- Thank you so much for this great project, please keep it alive!🤞

pyr0ball, 2024-09-25:
Sorry about the docs site, my home server suffered a massive hardware failure recently and I've been a little bogged down in recovering all of the webserver stuff. Thankfully all of the docs exist in this repo! 

https://github.com/pyr0ball/pyr0piezo/tree/master/docs

You can either read them directly using a markdown viewer, or by building and deploying with mkdocs

Parallel works with a single board! 

I still have a buttload of v2.3.2 boards floating around but all of my webserver stuff is toasted. If you really want one of these, we'll work something out.

If you don't need a compact version, you could build one yourself using the business card or mini variations. The board files for those are all in the repo as well if you want to source your own:

https://kitspace.org/boards/github.com/pyr0ball/pyr0piezo/pyr0-piezo%20z-probe%20standalone%20module/
https://kitspace.org/boards/github.com/pyr0ball/pyr0piezo/pyr0-piezo%20z-probe%20business%20card%20diy%20module/

hannpet, 2024-09-26:
Thanks für the reply! Too bad what happened to your server, that is really
painful.

So now I have read myself a little bit into the topic and got kind of
intrigued...
I think I'll try to hook up the Piezo-Rings to a breadboard with a rp2040
or esp32 and ask chatGpt to write me a MicroPython script to do the
sensor-autotuning.... it's probably going to be epic failure, but I kind of
feel like playing around a little bit 🙂

I'll definitely get back to you when I'm about to order a finished product,
thanks a lot Alan for the offer to supply one from your stack.

closed 2024-09-27 by hannpet

pyr0ball, 2024-11-20:
Just a heads up, the docs site is back online at https://docs.pyroballpcbs.com